The-OpenROAD-Project/RePlAce #6: Segmentation Fault when running example lefdef script

christosg88, 2018-11-15:
When executing the `execute_lefdef.py` script, the RePlAce executable starts OK, but when the LEF parser starts (I think), a segmentation fault occurs.

Here's the output of the command:
```
./execute_lefdef.py 0
  ./RePlAce -bmflag etc -lef ../bench/lefdef/ispd18_test1/ispd18_test1.input.lef -def ../bench/lefdef/ispd18_test1/ispd18_test1.input.def  -output ../output -t 1 -dpflag NTU3 -dploc ../ntuplace/ntuplace3   | tee ../logdir/ispd18_test1_2018-11-15_12:39:05.log 
CMD :  ./RePlAce -bmflag etc -lef ../bench/lefdef/ispd18_test1/ispd18_test1.input.lef -def ../bench/lefdef/ispd18_test1/ispd18_test1.input.def -output ../output -t 1 -dpflag NTU3 -dploc ../ntuplace/ntuplace3 
INFO:  VERSION, Compiled at Nov 15 2018 12:23:31
INFO:  Now is Thu Nov 15 12:39:05 2018
INFO:  Target Density = 1.000000
INFO:  Experiment Index 0
INFO:  DIR_PATH = ../output/etc/ispd18_test1.input/experiment0

Segmentation fault (core dumped)
```

mgwoo, 2018-11-27:
I've recorded this commit: https://github.com/abk-openroad/RePlAce/commit/da43841c94b38249b7d02d257d7e7b91cfc2bd77

Could you check this again?
p.s. I've created [this documents](https://github.com/abk-openroad/RePlAce/blob/master/doc/ReportMemoryBug.md). Could you report the _valgrind.log?

christosg88, 2018-11-29:
I can reproduce the issue, because latest commit doesn't compile. I'll open new issue and I'll come back once compilation succeeds.

christosg88, 2019-11-12:
I believe that the segmentation faults that keep happening, are because of the "-O3" optimization flag in the lef/def CMakeLists.txt.
Disabling the optimizations stops the segmentation faults.
I'm not sure which part of the parsers is responsible for the crash once the compiler optimizes it, but since the lef/def parsing is not a critical part of the project, I would suggest to just disable the optimizations for now, at least until SI2 fixes the problem.
I have created the relative pull requests for lef and def submodules, and also for the RePlace project.

https://github.com/The-OpenROAD-Project/lef/pull/4
https://github.com/The-OpenROAD-Project/def/pull/3
https://github.com/The-OpenROAD-Project/RePlAce/pull/57

closed 2019-11-15 by christosg88