The-OpenROAD-Project/RePlAce #7: installation issue

kunalg123, 2018-11-27:
Hi 
Getting below issue while installing RePlace. 

g++ -o RePlAce                               \
-std=c++0x -g -ggdb -m64 -O3 -fPIC -DNDEBUG -ffast-math -Dcimg_display=1  					    			\
/home/anaghavsd/Desktop/work/tools/RePlAce/src/timingSta.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/initPlacement.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/intelLibFs.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/bin.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/charge.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg2d.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/trial.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/gcell.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/plot.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/opt.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/myFunctions.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/tier.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/wlen.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/routeOpt.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/ns.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/macro.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg3d.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/detailPlace.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fft.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/argument.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/main.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/bookShelfIO.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/defParser.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefParser.o ../module/OpenSTA//app/StaApp_wrap.o ../module/OpenSTA//app/StaMain.o ../module/OpenSTA//app/TclInitVar.o 							\
 	-Wl,--start-group /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_intel_lp64.a /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_gnu_thread.a /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_core.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippi.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libipps.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippcore.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippvm.a ../module/OpenSTA/install-sp//lib/libliberty.a ../module/OpenSTA/install-sp//lib/libutil.a ../module/OpenSTA/install-sp//lib/libsearch.a ../module/OpenSTA/install-sp//lib/libdcalc.a ../module/OpenSTA/install-sp//lib/libgraph.a ../module/OpenSTA/install-sp//lib/libnetwork.a ../module/OpenSTA/install-sp//lib/libparasitics.a ../module/OpenSTA/install-sp//lib/libsdc.a ../module/OpenSTA/install-sp//lib/libsdf.a ../module/OpenSTA/install-sp//lib/libverilog.a -Wl,--end-group \
-lgomp -lpthread -lm -ldl -lX11 -ltcl8.5 -lz \
-L../module/lef/5.8-p027/lib -llef -llefzlib -lclef -lclefzlib -L../module/def/5.8-p027/lib -ldef -ldefzlib -lcdef -lcdefzlib -L../module/verilog-parser/lib -lverilog_parser -L../module/flute/lib -lflute -static-libstdc++ 	
/usr/bin/ld: cannot find -lverilog_parser
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target 'RePlAce' failed
make[1]: *** [RePlAce] Error 1
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

There are other missing dependencies which I was able to resolve manually, like tcl.h not found OR verilog_writer.h not found. But the above one, not sure how to proceed. 
Can you please help?

mgwoo, 2018-11-27:
Hi,

Did you follow the procedure in the README.md? 
([git --recursive] commend that clone the whole submodules?)

There must be a static-linked library as RePlAce/module/verilog-parser/lib/libverilog-parser.a

kunalg123, 2018-11-27:
Hi Mingyu
Yes, I am following steps given in README.md
I have noted all steps where it gets stuck

**First place where it gets stuck is below:**
In file included from /home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.cpp:2:0:
/home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.h:10:17: fatal error: tcl.h: No such file or directory
compilation terminated.
Makefile:84: recipe for target '/home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.o' failed
make[1]: *** [/home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.o] Error 1
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
So I go to src directory and do the below (I thought it should have linked to tcl automatically) :
anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$ **sudo ln -s /usr/include/tcl/*.h .**

**Second place where it gets stuck is below:**
g++ -std=c++0x -g -ggdb -m64 -O3 -fPIC -DNDEBUG -ffast-math -Dcimg_display=1  -I/home/anaghavsd/Desktop/work/tools/RePlAce/src -I/opt/intel/compilers_and_libraries/linux/mkl/include -I/opt/intel/compilers_and_libraries/linux/ipp/include -I../module/sparsehash/install-sp/include -I../module/eigen-git-mirror/ -I../module/CImg/ -I../module/lef/5.8-p027/include -I../module/def/5.8-p027/include -I../module/verilog-parser/include -I../module/flute/include -I../module/OpenSTA/install-sp//include -I../module/OpenSTA/ -c -o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.cpp 
/home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.cpp:52:28: fatal error: verilog_writer.h: No such file or directory
compilation terminated.
Makefile:84: recipe for target '/home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.o' failed
make[1]: *** [/home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.o] Error 1
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
I again go to src directory and do below (I thought this should be automated):
anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$ **sudo ln -s /home/anaghavsd/Desktop/work/tools/RePlAce/module/verilog-parser/src/*.h .**

**Third place where it gets stuck is the above place where I filed this issue:**
I am not finding the "verilog-parser.a". Infact I don't see that folder "lib". Am I missing anything??
anaghavsd@rise:~/Desktop/work/tools/RePlAce$ ls -altr module/verilog-parser/
total 64
drwxr-xr-x 10 root root 4096 Nov 27 09:10 ..
-rw-r--r--  1 root root   86 Nov 27 09:11 .git
-rw-r--r--  1 root root  814 Nov 27 09:11 .travis.yml
-rw-r--r--  1 root root 2496 Nov 27 09:11 README.md
-rw-r--r--  1 root root 1079 Nov 27 09:11 LICENSE.txt
-rw-r--r--  1 root root  186 Nov 27 09:11 .gitignore
drwxr-xr-x  2 root root 4096 Nov 27 09:11 .github
-rw-r--r--  1 root root 1505 Nov 27 09:11 CONTRIBUTING.md
-rw-r--r--  1 root root   27 Nov 27 09:11 .codedocs
drwxr-xr-x  3 root root 4096 Nov 27 09:11 module
drwxr-xr-x  2 root root 4096 Nov 27 09:11 docs
drwxr-xr-x  2 root root 4096 Nov 27 09:11 bin
-rw-r--r--  1 root root  639 Nov 27 09:11 scratchpad.md
drwxr-xr-x  8 root root 4096 Nov 27 09:11 .
drwxr-xr-x  2 root root 4096 Nov 27 09:11 tests
drwxr-xr-x  2 root root 4096 Nov 27 09:41 src

kunalg123, 2018-11-27:
One more comment
For the very first time, "make clean"  step gives the below error

make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/module/flute/src'
cd module/OpenSTA && make distclean && rm -rf install-sp > /dev/null 2>&1; true
make[1]: Entering directory '/home/anaghavsd/Desktop/work/tools/RePlAce/module/OpenSTA'
make[1]: *** No rule to make target 'distclean'.  Stop.
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/module/OpenSTA'
cd module/sparsehash && make distclean && rm -rf install;
make[1]: Entering directory '/home/anaghavsd/Desktop/work/tools/RePlAce/module/sparsehash'
make[1]: *** No rule to make target 'distclean'.  Stop.
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/module/sparsehash'
Makefile:29: recipe for target 'clean' failed
make: *** [clean] Error 2

mgwoo, 2018-11-27:
Could you give me the whole log when you type "$make" in "/RePlAce" folder? (not the "/RePlAce/src"?)
I was not able to find the compile log for module/verilog-parser modules...

kunalg123, 2018-11-28:
Hi 
Where is the log file created?
I am not seeing log file
kunalg@kunalg-VirtualBox ~/Desktop/work/tools/RePlAce $ ls -ltr 
total 48
-rw-r--r--  1 root root 2499 Nov 27 14:42 README.md
-rw-r--r--  1 root root  847 Nov 27 14:42 Makefile
-rw-r--r--  1 root root 1548 Nov 27 14:42 LICENSE
drwxr-xr-x  3 root root 4096 Nov 27 14:42 doc
drwxr-xr-x  4 root root 4096 Nov 27 14:42 bench
-rw-r--r--  1 root root  579 Nov 27 14:42 AUTHORS.md
drwxr-xr-x  2 root root 4096 Nov 27 14:42 logdir
drwxr-xr-x 10 root root 4096 Nov 27 14:42 module
drwxr-xr-x  2 root root 4096 Nov 27 14:42 ntuplace
drwxrwxrwx  2 root root 4096 Nov 27 14:42 prerequisite
drwxr-xr-x  2 root root 4096 Nov 27 14:42 output
drwxr-xr-x  2 root root 4096 Nov 27 17:14 src
kunalg@kunalg-VirtualBox ~/Desktop/work/tools/RePlAce $ find ./ -name "*.log*"
./module/OpenSTA/config.log
./module/sparsehash/config.log

kunalg123, 2018-11-28:
Hi Mingyu
Just wanted to confirm with that I am using below steps to install RePlace. Hope that is correct. Let me know if I am missing anything?

sudo wget https://apt.repos.intel.com/intel-gpg-keys/GPG-PUB-KEY-INTEL-SW-PRODUCTS-2019.PUB
sudo apt-key add GPG-PUB-KEY-INTEL-SW-PRODUCTS-2019.PUB
sudo wget https://apt.repos.intel.com/setup/intelproducts.list -O /etc/apt/sources.list.d/intelproducts.list
sudo sh -c 'echo deb https://apt.repos.intel.com/mkl all main > /etc/apt/sources.list.d/intel-mkl.list'
sudo sh -c 'echo deb https://apt.repos.intel.com/ipp all main > /etc/apt/sources.list.d/intel-ipp.list'
sudo dpkg --add-architecture i386
sudo apt-get update
**sudo apt-get install intel-mkl-2018.2-046
sudo apt-get install intel-ipp-2018.4-057**
sudo wget -O boost_1_55_0.tar.gz http://sourceforge.net/projects/boost/files/boost/1.55.0/boost_1_55_0.tar.gz/download
tar -xvzf boost_1_55_0.tar.gz
cd boost_1_55_0/
sudo apt-get update
sudo apt-get install libicu-dev build-essential libbz2-dev
sudo apt-get install build-essential  autotools-dev
sudo ./bootstrap.sh --prefix=/usr/local
sudo ./b2
cd ../
git clone --recursive https://github.com/abk-openroad/RePlAce.git
cd RePlAce
make clean (gives errors the first time)
./prerequisite/install_ubuntu.sh 
make (This is where it gives all above errors)

mgwoo, 2018-11-28:
It seems you're correct.
Could you generate your log files and upload here by using script commends? For example

    $ script output.log
    $ make
    $ exit

kunalg123, 2018-11-28:
[output.log](https://github.com/abk-openroad/RePlAce/files/2625009/output.log)
Log File attached

mgwoo, 2018-11-28:
Thanks for reporting this issue.
Could you check the [this commit](https://github.com/mgwoo/verilog-parser/commit/651b4c390c0e0c61abad53bcedcdf2ce21ac8b8f) and [this document](http://pubs.opengroup.org/onlinepubs/7908799/xcu/lex.html)?
This un-compilable issue comes from the compiler(flex) bug. (yyleng must be int)

Possible solutions are:
1) Install the newest version of flex (2.6.4 recommended), or 
2) Modify the line in RePlAce/module/verilog-parser/src/verilog_parser.h, as

        // from
        extern YY_BUFFER_STATE yy_scan_bytes (const char *bytes, int len);
        // to
        extern YY_BUFFER_STATE yy_scan_bytes (const char *bytes, yy_size_t len);

kunalg123, 2018-11-29:
The above worked, now it gets stuck at below step

g++ -o RePlAce                               \
-std=c++0x -g -ggdb -m64 -O3 -fPIC -DNDEBUG -ffast-math -Dcimg_display=1  					    			\
/home/anaghavsd/Desktop/work/tools/RePlAce/src/timingSta.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/initPlacement.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/intelLibFs.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/bin.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/charge.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg2d.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/trial.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefdefIO.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/gcell.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/plot.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/opt.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/myFunctions.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/tier.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/wlen.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/routeOpt.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/timing.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/ns.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/macro.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fftsg3d.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/detailPlace.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/fft.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/argument.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/main.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/bookShelfIO.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/defParser.o /home/anaghavsd/Desktop/work/tools/RePlAce/src/lefParser.o ../module/OpenSTA//app/StaApp_wrap.o ../module/OpenSTA//app/StaMain.o ../module/OpenSTA//app/TclInitVar.o 							\
 	-Wl,--start-group /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_intel_lp64.a /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_gnu_thread.a /opt/intel/compilers_and_libraries_2018.5.274/linux/mkl/lib/intel64/libmkl_core.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippi.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libipps.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippcore.a /opt/intel/compilers_and_libraries_2018.5.274/linux/ipp/lib/intel64/libippvm.a ../module/OpenSTA/install-sp//lib/libliberty.a ../module/OpenSTA/install-sp//lib/libutil.a ../module/OpenSTA/install-sp//lib/libsearch.a ../module/OpenSTA/install-sp//lib/libdcalc.a ../module/OpenSTA/install-sp//lib/libgraph.a ../module/OpenSTA/install-sp//lib/libnetwork.a ../module/OpenSTA/install-sp//lib/libparasitics.a ../module/OpenSTA/install-sp//lib/libsdc.a ../module/OpenSTA/install-sp//lib/libsdf.a ../module/OpenSTA/install-sp//lib/libverilog.a -Wl,--end-group \
-lgomp -lpthread -lm -ldl -lX11 -ltcl8.5 -lz \
-L../module/lef/5.8-p027/lib -llef -llefzlib -lclef -lclefzlib -L../module/def/5.8-p027/lib -ldef -ldefzlib -lcdef -lcdefzlib -L../module/verilog-parser/lib -lverilog_parser -L../module/flute/lib -lflute -static-libstdc++ 	
../module/OpenSTA//app/StaMain.o: In function `sta::staMain(sta::Sta*, int, char**, int (*)(Tcl_Interp*))':
StaMain.cc:(.text+0x95): undefined reference to `Tcl_MainEx'
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target 'RePlAce' failed
make[1]: *** [RePlAce] Error 1
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

kunalg123, 2018-11-29:
Attaching log file
[output1.log](https://github.com/abk-openroad/RePlAce/files/2627710/output1.log)

mgwoo, 2018-11-29:
Thanks for reporting this. 
Could you change the below line to have -ltcl8.6, instead of -ltcl8.5?

https://github.com/abk-openroad/RePlAce/blob/8b03ebe4da3f9d6b109664829859b9dfee1e3467/src/Makefile#L104

This'll work for final linking.
(I'm finding a good way that catches TCL library version...)

kunalg123, 2018-11-29:
Linking is complete, but next error is below
Attaching log
[output2.log](https://github.com/abk-openroad/RePlAce/files/2627986/output2.log)

  Linking is completed !!
 
  Tagging...
ctags -R *
/bin/sh: 1: ctags: not found
Makefile:92: recipe for target 'RePlAce' failed
make[1]: *** [RePlAce] Error 127
make[1]: Leaving directory '/home/anaghavsd/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
exit
Script done, file is output2.log

mgwoo, 2018-11-29:
ctags is optional. (Compile is finished)
You can ignore this.

kunalg123, 2018-11-29:
Alright
I am now running replace. While doing the below step, I am seeing errors. Anything missing?

// download lefdef benchmarks
$ cd ~/RePlAce/bench/lefdef
$ ./download_ispd18.sh

anaghavsd@rise:~/Desktop/work/tools/RePlAce/bench/lefdef$ ./download_ispd18.sh 
--2018-11-29 12:54:15--  http://www.ispd.cc/contests/18/ispd18_test%7B1..10%7D.tgz
Resolving www.ispd.cc (www.ispd.cc)... 64.238.159.114
Connecting to www.ispd.cc (www.ispd.cc)|64.238.159.114|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://archive.sigda.org/404.html [following]
--2018-11-29 12:54:16--  http://archive.sigda.org/404.html
Resolving archive.sigda.org (archive.sigda.org)... 64.238.159.114
Reusing existing connection to www.ispd.cc:80.
HTTP request sent, awaiting response... 200 OK
Length: 10586 (10K) [text/html]
Saving to: ‘ispd18_test{1..10}.tgz’

ispd18_test{1..10}.tgz                     100%[=====================================================================================>]  10.34K  24.1KB/s    in 0.4s    

2018-11-29 12:54:17 (24.1 KB/s) - ‘ispd18_test{1..10}.tgz’ saved [10586/10586]

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

kunalg123, 2018-11-29:
I think it should be 
for i in 1 2 3 4 5 6 7 8 9 10 
and not
for i in {1...10}

Can you please confirm?

mgwoo, 2018-11-29:
I guess it's not essential. 
1 2 3 4 5 ... and {1...10} worked well in Centos6/ubuntu16
Close Issue

closed 2018-11-29 by mgwoo

kunalg123, 2018-11-30:
Hi 
We tried to install RePlace on a different machine and it got stuck with below error. Log file attached. Anything missing?

  Building... /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o
g++ -std=c++0x -g -ggdb -m64 -O3 -fPIC -DNDEBUG -ffast-math -Dcimg_display=1  -I/home/kunalg/Desktop/work/tools/RePlAce/src -I/opt/intel/compilers_and_libraries/linux/mkl/include -I/opt/intel/compilers_and_libraries/linux/ipp/include -I../module/sparsehash/install-sp/include -I../module/eigen-git-mirror/ -I../module/CImg/ -I../module/lef/5.8-p027/include -I../module/def/5.8-p027/include -I../module/verilog-parser/include -I../module/flute/include -I../module/OpenSTA/install-sp//include -I../module/OpenSTA/ -c -o /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.cpp 
In file included from /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.cpp:1:0:
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:424:16: error: explicit specialization of ‘template<class _Tp> struct std::hash’ outside its namespace must use a nested-name-specifier [-fpermissive]
         struct hash< pair<DBU, DBU> > {
                ^~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:424:16: error: redefinition of ‘struct std::hash<std::pair<long int, long int> >’
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:401:16: note: previous definition of ‘struct std::hash<std::pair<long int, long int> >’
         struct hash< pair<DBU, DBU> > {
                ^~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:436:16: error: explicit specialization of ‘template<class _Tp> struct std::hash’ outside its namespace must use a nested-name-specifier [-fpermissive]
         struct hash< Timing::PinInfo > {
                ^~~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:436:16: error: redefinition of ‘struct std::hash<Timing::PinInfo>’
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:413:16: note: previous definition of ‘struct std::hash<Timing::PinInfo>’
         struct hash< Timing::PinInfo > {
                ^~~~~~~~~~~~~~~~~~~~~~~
Makefile:84: recipe for target '/home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o' failed
make[1]: *** [/home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o] Error 1
make[1]: Leaving directory '/home/kunalg/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
exit
Script done, file is output.log

kunalg123, 2018-11-30:
https://1drv.ms/u/s!Ai4WW_jutenggas7vxLc-aKr2B46pA