The-OpenROAD-Project/RePlAce #11: Cannot find SITE in lef - error

kunalg123, 2018-11-30:
Hi 

I am trying to do hierarchical placement and hence pointing to memory lef, pad lef, std cell lef and tech lef. 

anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$ ./execute_lefdef.py mkSoc_wrapper
  ./RePlAce -bmflag etc -lef ../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef -lef ../bench/lefdef/mkSoc_wrapper/DPRAM_32x8.lef -lef ../bench/lefdef/mkSoc_wrapper/tsl180l4.lef -lef ../bench/lefdef/mkSoc_wrapper/DPRAM_4096x36.lef -lef ../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x32.lef -lef ../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x8.lef -lef ../bench/lefdef/mkSoc_wrapper/tsl18fs120_scl.lef -def ../bench/lefdef/mkSoc_wrapper/mkSoc_wrapper.input.def  -output ../output -t 1 -dpflag NTU3 -dploc /home/anaghavsd/Desktop/work/tools/RePlAce/ntuplace/ntuplace3   | tee ../logdir/mkSoc_wrapper_2018-11-30_14:37:33.log 
CMD :  ./RePlAce -bmflag etc -lef ../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef -lef ../bench/lefdef/mkSoc_wrapper/DPRAM_32x8.lef -lef ../bench/lefdef/mkSoc_wrapper/tsl180l4.lef -lef ../bench/lefdef/mkSoc_wrapper/DPRAM_4096x36.lef -lef ../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x32.lef -lef ../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x8.lef -lef ../bench/lefdef/mkSoc_wrapper/tsl18fs120_scl.lef -def ../bench/lefdef/mkSoc_wrapper/mkSoc_wrapper.input.def -output ../output -t 1 -dpflag NTU3 -dploc /home/anaghavsd/Desktop/work/tools/RePlAce/ntuplace/ntuplace3 
INFO:  VERSION, Compiled at Nov 29 2018 12:22:24
INFO:  Now is Fri Nov 30 14:37:33 2018
INFO:  Target Density = 1.000000
INFO:  Experiment Index 2
INFO:  DIR_PATH = ../output/etc/mkSoc_wrapper.input/experiment002

PROC:  BEGIN IMPORTING PLACEMENT INPUT
WARNING (LEFPARS-2001): No VERSION statement found, using the default value 5.8. See file ../bench/lefdef/mkSoc_wrapper/DPRAM_4096x36.lef at line 23245.

** WARNING : LEF unit Info is missing..! (UNITS/DATABASE)
INFO:  DEF SCALE UNIT: 1000
INFO:  METAL1 NAME IN LEF: M1
INFO:  SCALE DOWN UNIT: ( 560, 560 )
INFO:  OFFSET COORDINATE: ( 160, 160 )

INFO:  #MODULE: 228718, #TERMINAL: 937

** ERROR:  Cannot find SITE in lef files: unit
-------------------------------------------------------
--------------------------------------------------------
But when I grep for SITE, I am seeing the below:
Am I missing anything?

anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$ grep SITE ../bench/lefdef/mkSoc_wrapper/*.lef | sort -u
../bench/lefdef/mkSoc_wrapper/DPRAM_32x8.lef:SITE CoreSite
../bench/lefdef/mkSoc_wrapper/DPRAM_32x8.lef:SITE IOSite_c
../bench/lefdef/mkSoc_wrapper/DPRAM_4096x36.lef:SITE CoreSite
../bench/lefdef/mkSoc_wrapper/DPRAM_4096x36.lef:SITE IOSite_c
../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x32.lef:SITE CoreSite
../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x32.lef:SITE IOSite_c
../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x8.lef:SITE CoreSite
../bench/lefdef/mkSoc_wrapper/obc11_dpram_256x8.lef:SITE IOSite_c
../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef:    SITE CO_MSTR_c ;
../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef:SITE CO_MSTR_c
../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef:    SITE IOSite_c ;
../bench/lefdef/mkSoc_wrapper/tsl18cio150_4lm.lef:SITE IOSite_c
../bench/lefdef/mkSoc_wrapper/tsl18fs120_scl.lef:    SITE CoreSite ;
../bench/lefdef/mkSoc_wrapper/tsl18fs120_scl.lef:SITE CoreSite
anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$

kunalg123, 2018-12-02:
Hi 
My DEF contains the below. 
I am dumping DEF for standard EDA tool, and looks like, there is no option to dump CoreSite. Can you please help how were you able to dump that info?

VERSION 5.8 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
DESIGN mkSoc_wrapper ;
UNITS DISTANCE MICRONS 2000 ;
DIEAREA ( 0 0 ) ( 22630000 22630000 ) ;
ROW STD_ROW_6669 unit 1020000 21421920 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6670 unit 1020000 21434240 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6671 unit 1020000 21446560 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6672 unit 1020000 21458880 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6673 unit 1020000 21471200 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6674 unit 1020000 21483520 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6675 unit 1020000 21495840 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6676 unit 1020000 21508160 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6677 unit 1020000 21520480 N DO 18383 BY 1 STEP 1120 0 ;
ROW STD_ROW_6678 unit 1020000 21532800 N DO 18383 BY 1 STEP 1120 0 ;

mgwoo, 2018-12-02:
I see...

Could you check that your previous flow contains "generating SITE [unit]"?
It seems the [unit] is somehow created by your previous tool.

Three possible solutions are
1) Fix the [unit] SITE creating issue in your previous tool (recommended), or 
2) If the width and height of your SITE on def is equal to SITE-CoreSite definition, manually replace 'unit' into other SITE, (e.g. CoreSite) on your def, or
3) Manually create unit SITE definition on your lef. For example

        SITE unit
            CLASS CORE;
            // the XXX and YYY must be calculated based on your generated def
            SIZE XXX BY YYY; 
        END unit

Close Issue.

closed 2018-12-02 by mgwoo