The-OpenROAD-Project/RePlAce #12: installation problem on new machine

kunalg123, 2018-11-30:
Hi
We tried to install RePlace on a different machine and it got stuck with below error. Log file attached. Anything missing?
https://1drv.ms/u/s!Ai4WW_jutenggas7vxLc-aKr2B46pA

Building... /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o
g++ -std=c++0x -g -ggdb -m64 -O3 -fPIC -DNDEBUG -ffast-math -Dcimg_display=1 -I/home/kunalg/Desktop/work/tools/RePlAce/src -I/opt/intel/compilers_and_libraries/linux/mkl/include -I/opt/intel/compilers_and_libraries/linux/ipp/include -I../module/sparsehash/install-sp/include -I../module/eigen-git-mirror/ -I../module/CImg/ -I../module/lef/5.8-p027/include -I../module/def/5.8-p027/include -I../module/verilog-parser/include -I../module/flute/include -I../module/OpenSTA/install-sp//include -I../module/OpenSTA/ -c -o /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.cpp
In file included from /home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.cpp:1:0:
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:424:16: error: explicit specialization of ‘template struct std::hash’ outside its namespace must use a nested-name-specifier [-fpermissive]
struct hash< pair<DBU, DBU> > {
^~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:424:16: error: redefinition of ‘struct std::hash<std::pair<long int, long int> >’
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:401:16: note: previous definition of ‘struct std::hash<std::pair<long int, long int> >’
struct hash< pair<DBU, DBU> > {
^~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:436:16: error: explicit specialization of ‘template struct std::hash’ outside its namespace must use a nested-name-specifier [-fpermissive]
struct hash< Timing::PinInfo > {
^~~~~~~~~~~~~~~~~~~~~~~
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:436:16: error: redefinition of ‘struct std::hashTiming::PinInfo’
/home/kunalg/Desktop/work/tools/RePlAce/src/timing.h:413:16: note: previous definition of ‘struct std::hashTiming::PinInfo’
struct hash< Timing::PinInfo > {
^~~~~~~~~~~~~~~~~~~~~~~
Makefile:84: recipe for target '/home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o' failed
make[1]: *** [/home/kunalg/Desktop/work/tools/RePlAce/src/timingSta.o] Error 1
make[1]: Leaving directory '/home/kunalg/Desktop/work/tools/RePlAce/src'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
exit
Script done, file is output.log

mgwoo, 2018-12-01:
I'm sorry, but Could you specify which OS did you use? (and GCC version?) 
I'd like to reproduce the same bugs on my side.

kunalg123, 2018-12-02:
Hi 
Attaching system info details and gcc version details
[image]
[image]