The-OpenROAD-Project/RePlAce #13: placement successful, but final def not created

kunalg123, 2018-12-02:
Hi 
I did what you suggested i.e. replace "unit" by "CoreSite" and RePlace now does takes these inputs and runs successfully. But at the output folder, I am not seeing final.def

Can you please check if placement ran correctly or not? Attaching log
[mkSoc_wrapper_2018-12-02_09:12:17.log](https://github.com/abk-openroad/RePlAce/files/2636716/mkSoc_wrapper_2018-12-02_09.12.17.log)

anaghavsd@rise:~/Desktop/work/tools/RePlAce/src$ ls -ltr ../output/etc/mkSoc_wrapper.input/experiment005/
total 13104
drwxrwxr-x 2 anaghavsd anaghavsd     4096 Dec  2 09:12 initPlace
drwxrwxr-x 2 anaghavsd anaghavsd     4096 Dec  2 09:12 cell
drwxrwxr-x 2 anaghavsd anaghavsd     4096 Dec  2 09:12 bin
drwxrwxr-x 2 anaghavsd anaghavsd     4096 Dec  2 09:12 arrow
-rw-rw-r-- 1 anaghavsd anaghavsd      288 Dec  2 11:36 mkSoc_wrapper.input_gp.spef
-rw-rw-r-- 1 anaghavsd anaghavsd 13396305 Dec  2 11:36 mkSoc_wrapper.input.eplace-mGP2D.pl

kunalg123, 2018-12-03:
Hi 
I again checked closely
Looks like placement stops giving below message. Attaching log. Can you please have a look?
[mkSoc_wrapper_yosys_synth_v1_2018-12-03_10:17:40.log](https://github.com/abk-openroad/RePlAce/files/2639248/mkSoc_wrapper_yosys_synth_v1_2018-12-03_10.17.40.log)

PROC:  BEGIN MACRO LEGALIZATION IN EACH TIER
PROC:  SA-based Macro Legalization (mLG)
INFO:  Boundary Bins Max_b_x=32, Max_b_y=32, Max_b_z=32
INFO:  Bin Step X=574.4688, Y=574.3750, Z=0.0312

mgwoo, 2018-12-05:
The result seems very weird to me...

1) Did you have an empty NET in DEF? 
RePlAce should take perfect netlist from DEF file (it seems some nets were empty, so it caused negative HPWL errors)
2) Did you intend to execute the mixed-size placement? (e.g. Do you have a multi-height cell as a 'PLACED' in  COMPONENTS section in DEF)? 

On the top of that, may I have the DEF file you've used? 
If I could get LEF/DEF file, this would be really perfect for me to fix this error. 
(If LEF is not allowed to share, I'll do my best to fix the errors based on the DEF file)

kunalg123, 2018-12-05:
[testcase.tar.gz](https://github.com/abk-openroad/RePlAce/files/2647906/testcase.tar.gz)
Hi 
I have recreated testcase with opensource std cells. So sharing the DEF and opensource std cells LEF in above attachment
Unfortunately the memories LEF's are foundary proprietary, so might not able to share them
And yes, I am using mixed-size placement

mgwoo, 2018-12-10:
I tried to re-generate your errors on my side, but I happened to know that memory name, memory size, and memory pin name is essential to run this benchmarks... It seems very hard to fix this issue....

closed.

closed 2018-12-10 by mgwoo