The-OpenROAD-Project/RePlAce #28: DEF parser does not recognize NAMECASESENSITIVE

RTimothyEdwards, 2018-12-23:
RePlAce fails when the DEF file contains the statement "NAMECASESENSITIVE ON ;"

mgwoo, 2018-12-25:
The NAMESCASESENSITIVE statement is only valid for LEF syntax.
I was not able to find this statement on DEF inside LEF/DEF 5.8 language manual... 
(RePlAce uses LEF/DEF parser and LEF/DEF language manual in this link: https://projects.si2.org/openeda.si2.org/project/showfiles.php?group_id=84#p145)

Could you check this again?

RTimothyEdwards, 2018-12-27:
Okay, never mind.  The keyword is "NAMESCASESENSITIVE" and I managed to spell it wrong.  Once spelled correctly, your parser flags it as obsolete in version 5.6, thus answering my other question.  So this is not an issue at all except for my own programming error, which I just fixed.

closed 2018-12-27 by RTimothyEdwards