The-OpenROAD-Project/RePlAce #29: Runtime errors

RTimothyEdwards, 2018-12-24:
Bear with me here, because I am trying to get RePlAce to work (1) in the context of qflow, and (2) as an executable installed in a standard install target, not with the executables contained within its own source directory tree.  So I have skipped right over the examples and plunged into running it on a pre-placement DEF file of my own creation (which is ongoing work, because I am still discovering what information RePlAce needs to see in the input DEF file).

So for a first run, I have managed to get through the first global placement phase.  I created an input DEF file and ran RePlAce with the following options (some of which I have very little understanding of, such as -bmflag and -dpflag):

   RePlAce -bmflag ispd -lef /usr/local/share/qflow/tech/osu035/osu35_stdcells.lef -def map9v3.def -output output -dpflag NTU3 -dploc output

Running this, it got all the way to "Call NtuPlacer3".  Just before it printed "Call NtuPlacer3", it output:

   ln -s output output/ispd/map9v3/experiment000/tiers/0/

I don't get that at all, as that appears to make a recursive symbolic link to itself.  Then the next thing it does is to do:

   cd output/ispd/map9v3/experiment000/tiers/0 && ./ntuplace3 -aux map9v3.aux -loadpl map9v3.pl -util 1.00 -noglobal

Which fails immediately, as there is no "ntuplace3" in that directory.  The indended "ntuplace3" is presumably the executable in the RePlAce source directory in ntuplace/ntuplace3.  What was the intention of this command?

RTimothyEdwards, 2018-12-27:
I actually got RePlAce to run to completion, and it appears to have created a reasonable placement of my digital block.  I did not need to use "-skipIP".  However, I was unable to get the placement to run with NTU4, only with NTU3.

tetra12, 2018-12-27:
@RTimothyEdwards  We committed a hello world test project in our PR with a standard input DEF, contributed by @kunalg123 .

The tool fails with a segm fault, but it's a good idea to run it first, since its input DEF was generated by a proven tool.

Replace crushes with Segmentation fault. Any ideas ?

RTimothyEdwards, 2018-12-27:
The original issue was the failure to understand the meaning of the -dploc argument.  This has since been fixed.  I may post another issue related to the failure to run NTU4, which is apparently a different problem, since I can get NTU3 to work correctly now.

closed 2018-12-27 by RTimothyEdwards

mgwoo, 2018-12-27:
Sorry for late reply.
The NTUplace4h requires *.route bookshelf file, but it is still not implemented yet. (so, It is only valid for Bookshelf benchmark). 
For commercial benchmark (LEF/DEF/Verilog), the NTUPlace3 is the only possible Detail Placer.

> The original issue was the failure to understand the meaning of the -dploc argument. This has since been fixed. I may post another issue related to the failure to run NTU4, which is apparently a different problem, since I can get NTU3 to work correctly now.