The-OpenROAD-Project/RePlAce #31: Revert verilog_parser.h last change

tetra12, 2018-12-27:
I couldn't open an issue in the corresponding repo ...
You should revert the last change in the verilog-parser repo, because it breaches linking for a 64-bit system:

--->> verilog_parser.h 	Following standard yy_scan_bytes(yy_size_t -> int) 	4 months ago

pls revert back to yy_size_t

mgwoo, 2018-12-27:
Hmm. Thanks for reporting this issue.
Is this really due to the 64/32bit environment problem? 
Then I could use some #define directive instead.

tetra12, 2018-12-27:
I don't know, but it surely doesn't work. Reported by 3 people around here

mgwoo, 2018-12-27:
Please look at the https://github.com/westes/flex/issues/118
If you see the compiled module/verilog-parser/src/verilog_scanner.c, you could discover "yyleng" as yy_size_t. It's a flex bug. (so, please update flex compiler. Default Ubuntu flex package seems to have this problem.)

You also could find the int yyleng definition in below.
http://pubs.opengroup.org/onlinepubs/7908799/xcu/lex.html 
It is defined as 'int'

mgwoo, 2018-12-27:
I happened to know that I've ported BISON, not the FLEX in the verilog-parser repository... I'll work on this to remove this kind of problem. 

If it is possible to avoid this problem by using define directive, it would be the best solution.

tetra12, 2018-12-27:
Mingue, I'd recommend you to focus on essential things. I get a **segmentation fault** for a hello world example, which you may see on your own in the /test  folder in my PR. And that's will be my next issue when you accept my changes from PR.

I personally lost quite a few time on this issue. This was report reported in #7 and #17 and people also lost their time for nothing. Some people around wanna try RePlAce in commercial projects, so lets focus on value added things, and leave Bison to its developers.

mgwoo, 2018-12-27:
the cell height is also wrong...
I've used below ROW statements

ROW STD_ROW_11 core 300 300 N DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_12 core 300 20300 FS DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_13 core 300 40300 N DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_14 core 300 60300 FS DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_15 core 300 80300 N DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_16 core 300 100300 FS DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_17 core 300 120300 N DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_18 core 300 140300 FS DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_19 core 300 160300 N DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_20 core 300 180300 FS DO 141 BY 1 STEP 1600 0 ;
ROW STD_ROW_21 core 300 200300 N DO 141 BY 1 STEP 1600 0 ;

so it generated below results:

[image]

mgwoo, 2018-12-28:
Both problems are solved. 
Close issue.

closed 2018-12-28 by mgwoo