The-OpenROAD-Project/RePlAce #33: Build failure fix

tetra12, 2019-01-04:
1. Current src/Makefile fails to build the project on Ubuntu 16. Pls find my patch attached.
TCL8.6 needs to be installed as a pre-requisite. 

https://drive.google.com/open?id=1FJAT2n-9OF0xw0niL_vZXebLeQE5-LiH

2. Also add uninstall target in ~/Makefile:
uninstall:
  rm -f $(INSTDIR)/RePlAce
  rm -f $(INSTDIR)/ntuplace*

mgwoo, 2019-01-11:
Thanks for reporting this issue.
1) The TCL 8.6 should be detected by cmake or another compile tool, so I'm finding a good way to catch this. The TCL version is really depending on environments. 
2) I'll add this in uninstall, but I guess ntuplace and FLUTE *.dat packages should not be in (INSTDIR)

mgwoo, 2019-05-29:
CMake has been implemented. Close.

closed 2019-05-29 by mgwoo