The-OpenROAD-Project/RePlAce #36: Floating point exception (core dumped)

kunalg123, 2019-04-11:
Hi 
I was running RePlACe for designs which we had uploaded. This time I am getting Segmentation fault. I did quick valgrind and below is what I get. Anything missing?

anaghavsd@rise:/scratch/work/my_tools/RePlAce/src$ valgrind /usr/local/bin/RePlAce -bmflag etc -lef ../bench/lefdef/mksoc_wrapper/DPRAM_32x8.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x8.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18cio150_4lm.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18fs120_scl.lef -lef ../bench/lefdef/mksoc_wrapper/tsl180l4.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x32.lef -lef ../bench/lefdef/mksoc_wrapper/DPRAM_4096x36.lef -def ../bench/lefdef/mksoc_wrapper/mkSoc_wrapper.floorplan.def  -output ../output -t 1 -dpflag NTU3 -dploc /scratch/work/my_tools/RePlAce/ntuplace/ntuplace3
==15572== Memcheck, a memory error detector
==15572== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==15572== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==15572== Command: /usr/local/bin/RePlAce -bmflag etc -lef ../bench/lefdef/mksoc_wrapper/DPRAM_32x8.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x8.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18cio150_4lm.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18fs120_scl.lef -lef ../bench/lefdef/mksoc_wrapper/tsl180l4.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x32.lef -lef ../bench/lefdef/mksoc_wrapper/DPRAM_4096x36.lef -def ../bench/lefdef/mksoc_wrapper/mkSoc_wrapper.floorplan.def -output ../output -t 1 -dpflag NTU3 -dploc /scratch/work/my_tools/RePlAce/ntuplace/ntuplace3
==15572== 
CMD :  /usr/local/bin/RePlAce -bmflag etc -lef ../bench/lefdef/mksoc_wrapper/DPRAM_32x8.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x8.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18cio150_4lm.lef -lef ../bench/lefdef/mksoc_wrapper/tsl18fs120_scl.lef -lef ../bench/lefdef/mksoc_wrapper/tsl180l4.lef -lef ../bench/lefdef/mksoc_wrapper/obc11_dpram_256x32.lef -lef ../bench/lefdef/mksoc_wrapper/DPRAM_4096x36.lef -def ../bench/lefdef/mksoc_wrapper/mkSoc_wrapper.floorplan.def -output ../output -t 1 -dpflag NTU3 -dploc /scratch/work/my_tools/RePlAce/ntuplace/ntuplace3 
INFO:  VERSION, Compiled at Apr 11 2019 10:20:31
INFO:  Now is Thu Apr 11 11:48:24 2019
INFO:  Target Density = 1.000000
INFO:  Experiment Index 5
INFO:  DIR_PATH = ../output/etc/mkSoc_wrapper.floorplan/experiment005

PROC:  BEGIN IMPORTING PLACEMENT INPUT
INFO:  DEF SCALE UNIT: 1000
INFO:  METAL1 NAME IN LEF: M1
==15572== Invalid read of size 4
==15572==    at 0x1BF9AE: SetParameter() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C17E1: ParseLefDef() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C1AAC: ParseInput() (in /usr/local/bin/RePlAce)
==15572==    by 0x18E2DF: main (in /usr/local/bin/RePlAce)
==15572==  Address 0x71935e0 is 32 bytes before a block of size 32 in arena "client"
==15572== 
==15572== Invalid read of size 8
==15572==    at 0x2AD7AB0: LefDefParser::lefiSite::sizeY() const (lefiMisc.cpp:1365)
==15572==    by 0x1BF9D3: SetParameter() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C17E1: ParseLefDef() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C1AAC: ParseInput() (in /usr/local/bin/RePlAce)
==15572==    by 0x18E2DF: main (in /usr/local/bin/RePlAce)
==15572==  Address 0x84142d8 is 24 bytes before a block of size 8 alloc'd
==15572==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15572==    by 0x2AC9273: LefDefParser::lefiPin::Init() (lefiMacro.cpp:788)
==15572==    by 0x2AC9560: LefDefParser::lefiPin::lefiPin(LefDefParser::lefiPin const&) (lefiMacro.cpp:931)
==15572==    by 0x29A0EE: void std::vector<LefDefParser::lefiPin, std::allocator<LefDefParser::lefiPin> >::_M_realloc_insert<LefDefParser::lefiPin const&>(__gnu_cxx::__normal_iterator<LefDefParser::lefiPin*, std::vector<LefDefParser::lefiPin, std::allocator<LefDefParser::lefiPin> > >, LefDefParser::lefiPin const&) (in /usr/local/bin/RePlAce)
==15572==    by 0x2983E5: pinCB(LefDefParser::lefrCallbackType_e, LefDefParser::lefiPin*, void*) (in /usr/local/bin/RePlAce)
==15572==    by 0x2B1669A: LefDefParser::lefyyparse() (lef.y:4685)
==15572==    by 0x295A48: Circuit::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (in /usr/local/bin/RePlAce)
==15572==    by 0x1C1785: ParseLefDef() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C1AAC: ParseInput() (in /usr/local/bin/RePlAce)
==15572==    by 0x18E2DF: main (in /usr/local/bin/RePlAce)
==15572== 
INFO:  SCALE DOWN UNIT: ( 0, 0 )
==15572== 
==15572== Process terminating with default action of signal 8 (SIGFPE)
==15572==  Integer divide by zero at address 0x1004096A32
==15572==    at 0x1BFB02: SetParameter() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C17E1: ParseLefDef() (in /usr/local/bin/RePlAce)
==15572==    by 0x1C1AAC: ParseInput() (in /usr/local/bin/RePlAce)
==15572==    by 0x18E2DF: main (in /usr/local/bin/RePlAce)
==15572== 
==15572== HEAP SUMMARY:
==15572==     in use at exit: 1,020,189,688 bytes in 18,419,090 blocks
==15572==   total heap usage: 44,575,167 allocs, 26,156,077 frees, 76,521,484,891 bytes allocated
==15572== 
==15572== LEAK SUMMARY:
==15572==    definitely lost: 272,998,198 bytes in 7,471,471 blocks
==15572==    indirectly lost: 7,429,786 bytes in 705,319 blocks
==15572==      possibly lost: 548 bytes in 7 blocks
==15572==    still reachable: 739,761,156 bytes in 10,242,293 blocks
==15572==         suppressed: 0 bytes in 0 blocks
==15572== Rerun with --leak-check=full to see details of leaked memory
==15572== 
==15572== For counts of detected and suppressed errors, rerun with: -v
==15572== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)

Floating point exception (core dumped)

kunalg123, 2019-04-11:
Hi
You can use below testcase:
[testcase.tar.gz](https://github.com/abk-openroad/RePlAce/files/3068451/testcase.tar.gz)
Use the below command:
valgrind RePlAce -bmflag etc -lef /usr/local/share/qflow/tech/osu018/osu018_stdcells.lef  -def layout/map9v3.def -output replace_output

Thanks
Kunal

mgwoo, 2019-04-13:
Hi Kunal,
Your DEF don't have ROW definition, so it's not possible to generate a placement solution. 
I'll generate an error for this.

kunalg123, 2019-04-14:
Thanks Mingyu. Got it...
Do you have any script that would generate crude floorplan def (with the right syntax) from synthesized netlist? If yes, can you please share?

mgwoo, 2019-04-14:
We're building that program now.  (called Verilog to DEF translator/generator). This generator is required to execute both of floorplanner and placer. It'll be open-sourced soon.

kunalg123, 2019-04-14:
Sounds good 
Thanks Mingyu

mgwoo, 2019-04-15:
Welcome.

closed 2019-04-15 by mgwoo