The-OpenROAD-Project/RePlAce #37: Source changes needed for build on Fedora

RTimothyEdwards, 2019-05-03:
I have a Fedora system, and I needed to make the following changes to RePlaCe to get it to compile (more or less) cleanly.  I do not know if these changes are Fedora-specific or not.

(1) src/global.h:  in structures MODULE, TERM, and NET, each declares a field "char name[255]" and initializes it with 'name("")'.  This causes my compiler to flag an error complaining about the difference between const char width 1 and const char width 255.  I corrected it by replacing the initialization with "name({'\0'})".

(2) src/timingSta.cpp:  My version of gcc did not recognize "std::defaultfloat" (line 312), and as far as I can tell, removing this line ("cout << std::defaultfloat;") has no impact on the output.

I believe that I have reported these before in another thread.  As simple changes that make the code compile on a larger number of systems, I recommend them.

---Tim

mgwoo, 2019-05-03:
Hi Tim, 
Thanks for reporting these issues.

(1) I'm trying to remove "char name[255]" in all of the structure. I think It's nonsense to reserve a huge memory space on the frequently-used structure. 
(2) Ok. I'll remove it now. 

Thanks,

mgwoo, 2019-05-03:
Updated!

closed 2019-05-03 by mgwoo