The-OpenROAD-Project/RePlAce #39: Pin constraints and placement?

RTimothyEdwards, 2019-05-08:
I can't find any way to specify pin constraints, and RePlAce appears to leave any pins in the PLACED position they started in.  If they are in the DEF file but do not have a PLACED position, they don't show up in the layout at all.  Pin constraints in, say, Cadence Encounter are passed through a Tcl setup file, but I can't find anything similar for RePlAce.

Is RePlAce able to optimize pin positions?  If so, what do I need to do to enable that?

---Tim

mgwoo, 2019-05-15:
Hi Tim,

RePlAce has no ability to place pins now. But, the pin-placement project is ongoing with UFRGS. 
I guess RePlAce with pin-placer will be updated soon.

Kind regards,
Mingyu Woo

mgwoo, 2019-07-22:
Pin Placer has been updated in [The-OpenROAD-Project/ioPlacer](https://github.com/The-OpenROAD-Project/ioPlacer) and ioPlacer performs iterative runs in binary mode (ioPlacer -> replace -> ioPlacer -> ...).
Close.

closed 2019-07-22 by mgwoo