The-OpenROAD-Project/RePlAce #41: Bookshelf Format Support

ArEsKay3, 2019-05-28:
Thanks for open sourcing this guys! Really helpful work.
Can someone help me get moving with the bookshelf format of the DAC2012 superblue benchmarks?
https://github.com/abk-openroad/RePlAce/commit/c02698e016144d806f59ed833a59cddd94515a07
It looks like support was removed for these benchmarks and I'd love to play around with the routability driven placement portion of this work.

mgwoo, 2019-05-28:
Hi ArEsKay3,

You can find the RePlAce that supporting bookshelf on the following link:
https://github.com/abk-openroad/RePlAce/tree/1.1.0
https://github.com/abk-openroad/RePlAce/releases/tag/1.1.0

and this is the last version that supporting bookshelf.
(This version contains DAC2012 benchmark as an example run.)

We choose not to support bookshelf anymore because academic format(bookshelf) has lots of limitations in the real world placement. As you know, Bookshelf format can't provide information like macro-HALO, rectilinear DIEAREA, fragmented-ROW shape, irregular metal layer pitch, PDN metal, and PDN location, etc...

But, commercial format (e.g. LEF/DEF) can already handle these limitations. I'm still trying to enable these things, but I've faced with huge brittleness of RePlAce because all of the RePlAces' structures are based on the simple, research-oriented format (bookshelf).

Hence, We've concluded that we'll not support bookshelf anymore. 
I hope this answer could help you.

Thanks,
-- mgwoo

ArEsKay3, 2019-05-29:
Thanks mgwoo!

I've been able to build and run the `test/run.sh` at this tag, but I get the below error when running.
`./execute_bookshelf.py superblue19`
`ERROR: RouteInst failed to get LAYER information from LEF`

Do you know what I might be missing?

mgwoo, 2019-05-29:
Oops, Sorry about this. 
Could you try again with the previous version? (e.g. 1.0.0 or 1.0.1 or initial commit?)
I think that should be fine for bookshelf formats. Somehow, I'll modify the description on 1.1.0.

ArEsKay3, 2019-05-29:
Thanks, I'll give that a shot.
So none of the bookshelf formats will work with your timing driven features? Or do they exist at 1.0.1? Just curious to look at that too.
Which are the benchmarks you are using which use lef/def in case I wanted to look at those instead?

mgwoo, 2019-05-29:
The bookshelf format doesn't have any timing information, so timing-driven placement in bookshelf format is impossible. You should have *.lef, *.def, *.lib, *.sdc, and *.v to execute Timing-Driven Placement.
(Timing-info is on *.lib and *.sdc)

To check the timing-driven engine, I already updated one benchmark (wb_dma_top) in the following link:
https://github.com/abk-openroad/RePlAce/tree/master/test/a2a_wb_dma_top

ArEsKay3, 2019-05-29:
Oh, ok. I hadn't even looked at that yet, but that makes sense (wrt timing and bookshelf files).

I can get 1.0.0 to build but run on the ICCAD benchmarks. I had to link my bookshelf dir to `src/SB_orig` and am still getting errors related to a `convert1.tcl` file.
```
INFO:  Temp. placement solution for Global Router has been written at
       ../output/etc/superblue19/experiment4/router/tier0/inflarion_iter0/superblue19.pl

sh: 1: convert1.tcl: not found
ERROR: ERROR: Cannot open: ../output/etc/superblue19/experiment4/router/tier0/inflarion_iter0/superblue19.est file
```

(the below is FYI if anyone else stumbles on this)

I can't get 1.0.1 to build (there seems to be a problem with 'char const *' or 'const char *' in the verilog-parser module at that tag, maybe related to https://github.com/abk-openroad/RePlAce/issues/17#issuecomment-449868976 ).
```
g++ -g -gstrict-dwarf -gdwarf-2 -m64 -O3 -fPIC -fno-strict-aliasing -fexceptions -DIL_STD -Wno-ctor-dtor-privacy -fno-permissive verilog_ast.o verilog_ast_common.o verilog_ast_mem.o verilog_ast_util.o verilog_parser.tab.o verilog_parser_wrapper.o verilog_preprocessor.o verilog_scanner.o verilog_writer.o main.o -o parser -O3 -fPIC -g -ggdb -fno-permissive -static-libstdc++
verilog_parser_wrapper.o: In function `verilog::verilog_parse_string(char*, int)':
/opt/RePlAce/module/verilog-parser/src/verilog_parser_wrapper.c:49: undefined reference to `yy_scan_bytes(char const*, int)'
```

And in order to get 1.1.0 to build I had to do the following.
```
# Fix for locating tcl.h and linking libtcl8.6
# (the first part might be because I have 8.6 instead of 8.5, or because I'm on Ubuntu16, idk)
 sed -i 's!tcl.h!tcl/tcl.h!g' src/timingSta.cpp
 sed -i 's!tcl.h!tcl/tcl.h!g' src/timing.h
 sed -i 's!tcl.h!tcl/tcl.h!g' module/OpenSTA/util/Report.hh
 sed -i 's!ltcl8.5!ltcl8.6!g' src/Makefile

# Fix .jpg issues (the second line is a bit of a hack to find the proper linker line in the makefile)
# This might be because I didn't use the scripts in prerequisites directory,
# I just installed everything manually
 gawk -i inplace '/CImg.h/ { print "#define cimg_use_jpeg"; print; next}1' src/plot.cpp
 sed -i 's!llef !llef -ljpeg !g' src/Makefile```

mgwoo, 2019-05-29:
I think you don't have to compile previous versions. 
I uploaded a compiled binary on [the release section](https://github.com/abk-openroad/RePlAce/releases).

I think you found the correct path in 1.0.0. 
Could you try with the attached zip file? I hope this file contains tcl files that you're looking for.
[tcls.zip](https://github.com/abk-openroad/RePlAce/files/3234743/tcls.zip)
(I found these from a folder that had been created a long time ago.)

For issues in 1.0.1 and 1.1.0, the current RePlAce doesn't have these problems anymore.
(I've removed Verilog-parser dependency, and RePlAce is compiled on CMake now.)

ArEsKay3, 2019-05-31:
I do if I want to play around with the code! :)

The attached zip file solved my problem and I'm able to run global route now on bookshelf designs!

I'm aware these issues are fixed now, just wanted to post in case anyone wants to go back to the bookshelf benchmarks like I did.

Thanks so much for the quick and helpful replies, I'm all set for now.

- ArEsKay3

closed 2019-05-31 by ArEsKay3