The-OpenROAD-Project/RePlAce #43: RePlAce portability for OpenROAD

tajayi, 2019-06-01:
I have a few suggestions to make RePlAce a little more portable, at least for OpenROAD project.
I like the new cmake system but it is statically linking even less things..

1. Can you please update the cmake process to statically link more libraries. At least the following were staticallly linked before: libICE.so.6, libmkl_core.so, libmkl_gnu_thread.so, libmkl_intel_lp64.so, libSM.so.6, libstdc++.so.6, libuuid.so.1, libXext.so.6

1. Can you also add "cimg_use_jpeg" to the CImg and statically link libjpeg? I don't use the "-plot" options in my test because it crashes looking for libjpeg (not installed on our servers)

Thanks

mgwoo, 2019-06-04:
Hi Tutu,
Sorry for the late reply. I've started to work on the detailed placer recently.
1. The FindMKL.cmake is from PyTorch. The default setting is to find a shared library instead of a static library. I'm fixing this issue now.
2. I'll add this to CMake.

tajayi, 2019-06-05:
Currently experiencing a build failure
```
CMake Error at CMakeLists.txt:46 (add_subdirectory):
  add_subdirectory given source "/RePlAce/module/ioPlacer" which is not an
  existing directory.
```

I'm guessing a new submodule hasn't been added to git?

tajayi, 2019-06-05:
Also 
```
CMake Error at CMakeLists.txt:59 (find_package):
  By not providing "Findjpeg.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "jpeg", but
  CMake did not find one.

  Could not find a package configuration file provided by "jpeg" with any of
  the following names:

    jpegConfig.cmake
    jpeg-config.cmake

  Add the installation prefix of "jpeg" to CMAKE_PREFIX_PATH or set
  "jpeg_DIR" to a directory containing one of the above files.  If "jpeg"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

I have libjpeg-turbo-static installed from yum and /usr/lib64/libjpeg.a is present.

Thanks

mgwoo, 2019-06-05:
> Currently experiencing a build failure
> 
> ```
> CMake Error at CMakeLists.txt:46 (add_subdirectory):
>   add_subdirectory given source "/RePlAce/module/ioPlacer" which is not an
>   existing directory.
> ```
> 
> I'm guessing a new submodule hasn't been added to git?

Oops, Sorry. I'm porting ioPlacer into RePlAce and it messed up. I'll solve this error.

> Also
>
> ```
> CMake Error at CMakeLists.txt:59 (find_package):
>   By not providing "Findjpeg.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by [...]

Can you tell me the CMake version that you're using? It seems your CMake doesn't support 

    find_package(jpeg)

 command.

tajayi, 2019-06-05:
> Can you tell me the CMake version that you're using? It seems your CMake doesn't support
cmake version 3.9.0

mgwoo, 2019-06-11:
I've updated the repo. Could you check this again?

tajayi, 2019-06-12:
Validated resolved... At least it runs on michigan and ucsd servers. The name of the executable changed from RePlAce to replace. Is that intentional?
Thanks

closed 2019-06-12 by tajayi