The-OpenROAD-Project/RePlAce #46: RePlAce + OpenDP has several placement violations

tajayi, 2019-06-19:
Running OpenDP after Replace yeilds several 

```
## Total instances with placement violations: 1393
## Insts placed overlapping with other insts (671) ##
### 722 Orientation Violation>
## Number of Placed Instances = 9694
## Number of Unplaced Instances = 0
## Placement Density:39.99%(29849/74650)
```

I've added t9_opendp_2019-06-19_15-22.tar.gz to the attachments

mgwoo, 2019-06-19:
Hi Tutu, 
It seems that your test case is very extreme cases.. (Utilization is 100%, and it's not realistic)
Can you modify your design to have more enough spaces in the floorplan step? I think 50-80% would be the usual case.

tajayi, 2019-06-19:
Good catch. This was as a result of the "lef padding". It appears to take the density of aes from about 40% to 102%. I reduced the padding from 2 microns to 1 micron (now about 70%). This still leaves an addition 5 routing tracks to the right of each cell so I think it should be sufficient.

closed 2019-06-19 by tajayi