The-OpenROAD-Project/RePlAce #49: Segfault in test-1-large-pin-count test-case

oharboe, 2019-08-09:
To reproduce:

Get this pull request for alpha-release: https://github.com/The-OpenROAD-Project/alpha-release/pull/28

Then run:

```
# cd alpha-release/flow/
# make DESIGN_CONFIG=designs/test-1-large-pin-count.mk
[deleted]
INFO:  Now linking: LargePinCount
INFO:  Successfully linked: LargePinCount
NO PATH !
INFO: Total 0 worst path!
/bin/bash: line 17:    69 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./objects/nangate45/LargePinCount/merged_padded.lef -def ./results/nangate45/LargePinCount/2_floorplan.def -verilog ./results/nangate45/LargePinCount/2_floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/nangate45/LargePinCount/2_floorplan.sdc -output ./results/nangate45/LargePinCount/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
        70 Done                    | tee ./logs/nangate45/LargePinCount/3_1_RePlAce.log

```

oharboe, 2019-08-09:
Similarly, this test fails:

```
# make DESIGN_CONFIG=designs/test-2-large-pin-count-some-meat.mk
INFO: Total 0 worst path!
/bin/bash: line 17:   139 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./objects/nangate45/LargePinCount/merged_padded.lef -def ./results/nangate45/LargePinCount/2_floorplan.def -verilog ./results/nangate45/LargePinCount/2_floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/nangate45/LargePinCount/2_floorplan.sdc -output ./results/nangate45/LargePinCount/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
       140 Done                    | tee ./logs/nangate45/LargePinCount/3_1_RePlAce.log
make: *** [results/nangate45/LargePinCount/3_1_place_gp.def] Error 139
```

oharboe, 2019-08-09:
Similarly, this test test fails:

```
# make DESIGN_CONFIG=designs/test-3-small-pin-count.mk
[deleted]
NO PATH !
INFO: Total 0 worst path!
/bin/bash: line 17:   207 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./objects/nangate45/SmallPinCount/merged_padded.lef -def ./results/nangate45/SmallPinCount/2_floorplan.def -verilog ./results/nangate45/SmallPinCount/2_floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/nangate45/SmallPinCount/2_floorplan.sdc -output ./results/nangate45/SmallPinCount/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
       208 Done                    | tee ./logs/nangate45/SmallPinCount/3_1_RePlAce.log
make: *** [results/nangate45/SmallPinCount/3_1_place_gp.def] Error 139
```

Sora-dcliu, 2019-08-30:
Excuse me， have u solved this problem?  I met a problem like this too.

christosg88, 2019-11-15:
Please try again with tip of master.
There was a bug fix for lef/def parser.

oharboe, 2019-11-15:
manual retesting of submitted test cases make no sense, it willjust break
with some future commit. this has to be automated. meanwhile, it doesnt
make much of a difference if it is retested or not.

i have posted an issue to install an automated pull request pre merge
policy, really it is two for the par these days, and i understand it is
openroad's intention to step up the game to do this.

fre. 15. nov. 2019, 13:36 skrev Christos Gantidis <notifications@github.com

christosg88, 2019-11-15:
The thing is that this was caused only with later version of gcc from what I can tell.

From the requirements at the home page I see 

> GCC compiler and libstdc++ static library >= 4.8.5

So if gcc4.8.5 didn't produce this behavior, would the automated test catch it?

Is there a way to define the tests to be run with binaries compiled with a set of gcc versions?

oharboe, 2019-11-15:
Good points. I guess what one can do is to test automtically one or more
environments, welcome pull requests that look good and address other
environments, but dot break the automqtically tested environments.

fre. 15. nov. 2019, 15:12 skrev Christos Gantidis <notifications@github.com