The-OpenROAD-Project/RePlAce #52: Compile issue for ppc64le.

iam41xin, 2019-08-15:
While compiling RePlAce under ppc64le, I've met an issue that components 'intel-mkl-2018.2-046' and 'intel-ipp-2018.4-057' were not available for ppc64le.
Would you mind my asking what's the purpose of using Intel-math-kernel here? Are there any equivalent components suitable for ppc64le?
Thanks!

mgwoo, 2019-08-23:
I've found out that RePlAce uses mkl just to use "mkl_malloc" function.
I guess if you replace mkl_malloc into malloc, then you could erase mkl.h include.
Removing MKL dependency is planned, so I'll do later.

iam41xin, 2019-08-23:
> I've found out that RePlAce uses mkl just to use "mkl_malloc" function.
> I guess if you replace mkl_malloc into malloc, then you could erase mkl.h include.

Thanks you very much for your valuable information, I tried to use OpenBLAS to replace mkl but I don't know which head files should be included. I'll try your solution to compile again soon!

iam41xin, 2019-08-26:
I've finished to compile RePlAce under ppc64le by using following methods, I don't know if it's the right way to solve this issue, just for reference. I'm still taking the function test.
1. exclude mkl.h and include malloc.h;
2. replace mkl_malloc with malloc and remove ', 64' after malloc in code;
3. replace mkl_free with free;
4. compile and link with -fopenmp to enable OpenMP. (set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Dcimg_use_jpeg=1 -O3 -fopenmp") in CMakeLists.txt)

iam41xin, 2019-08-27:
There are some special cases like this:
moduleInstance=(struct MODULE*)malloc(sizeof(struct
MODULE)*moduleCNT,64);
'malloc' and ',64' are not in the same line.

iam41xin, 2019-08-27:
Thank you very much, I can close this issue now.

closed 2019-08-27 by iam41xin

mgwoo, 2019-08-29:
I somehow disabled some realloc function to avoid memory violations in the 'dev' branch.
(Behaviors of mkl_realloc and realloc seem different, so realloc had a problem.)

After my current implementation for TCL command is finished, I would merge the 'dev' branch into 'master' branch.