The-OpenROAD-Project/RePlAce #53: [Compile Error] src/timing.h:431:27: error: ‘DBU’ was not declared in this scope

speedbooster, 2019-08-21:
Hi, I'm getting the following error while compiling with GCC-4.8.5 and GCC-8.3.0 on Centos7.

`In file included from /opt/intel/mkl/include/mkl.h:31:0,
                 from /home/user/Downloads/openflow/RePlAce/src/lefdefIO.cpp:53:
/opt/intel/mkl/include/mkl_cblas.h:916:46: warning: redundant redeclaration of ‘mkl_jit_status_t mkl_jit_destroy(void*)’ in same scope [-Wredundant-decls]
 mkl_jit_status_t mkl_jit_destroy(void* jitter);
                                              ^
In file included from /opt/intel/mkl/include/mkl.h:29:0,
                 from /home/user/Downloads/openflow/RePlAce/src/lefdefIO.cpp:53:
/opt/intel/mkl/include/mkl_blas.h:1059:18: warning: previous declaration of ‘mkl_jit_status_t mkl_jit_destroy(void*)’ [-Wredundant-decls]
 mkl_jit_status_t mkl_jit_destroy(void* jitter);
                  ^
In file included from /home/user/Downloads/openflow/RePlAce/src/lefdefIO.cpp:55:0:
/home/user/Downloads/openflow/RePlAce/src/timing.h:431:27: error: ‘DBU’ was not declared in this scope
 struct MyHash< std::pair< DBU, DBU > > {
                           ^
/home/user/Downloads/openflow/RePlAce/src/timing.h:431:32: error: ‘DBU’ was not declared in this scope
 struct MyHash< std::pair< DBU, DBU > > {
                                ^
/home/user/Downloads/openflow/RePlAce/src/timing.h:431:36: error: template argument 1 is invalid
 struct MyHash< std::pair< DBU, DBU > > {
                                    ^
/home/user/Downloads/openflow/RePlAce/src/timing.h:431:36: error: template argument 2 is invalid
/home/user/Downloads/openflow/RePlAce/src/timing.h:431:38: error: template argument 1 is invalid
 struct MyHash< std::pair< DBU, DBU > > {
`

I did not run the prerequisites/install script as I already installed the dependencies. I did run **yum install libstdc++-devel** and **libstdc++-static**.

Here's the output for **cmake ..** in build directory.
`
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found BISON: /usr/bin/bison (found version "3.0.4") 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.7") 
-- leflib_home: /home/user/Downloads/openflow/RePlAce/module/lef
-- Flute version: 3.3
-- System name: Linux
-- Compiler: GNU 4.8.5
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3
-- CXX_FLAGS: 
-- Install prefix: /usr/local
-- STA version: 2.0.16
-- STA git sha: 73fef1117e973efde68b1d39c6adabf5c83f6417
-- System name: Linux
-- Compiler: GNU 4.8.5
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3
-- STA CXX_FLAGS: -Wall;-Wcast-qual;-Wunused-parameter;-Wredundant-decls
-- Install prefix: /usr/local
-- CUDD library: not found
-- SSTA: 0
-- TCL lib: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found FLEX: /usr/bin/flex (found version "2.5.37") 
-- Found SWIG: /usr/bin/swig (found version "2.0.10") 
-- STA executable: /home/user/Downloads/openflow/RePlAce/module/OpenSTA/app/sta
-- Could NOT find GOOGLEHASH (missing: GOOGLEHASH_INCLUDES GOOGLEHASH_COMPILE) 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of void*
-- Check size of void* - done
-- Checking for [mkl_intel_lp64 - mkl_gnu_thread - mkl_core - gomp - pthread - m - dl]
--   Library mkl_intel_lp64: /opt/intel/mkl/lib/intel64/libmkl_intel_lp64.a
--   Library mkl_gnu_thread: /opt/intel/mkl/lib/intel64/libmkl_gnu_thread.a
--   Library mkl_core: /opt/intel/mkl/lib/intel64/libmkl_core.a
--   Library gomp: -fopenmp
--   Library pthread: /usr/lib64/libpthread.so
--   Library m: /usr/lib64/libm.so
--   Library dl: /usr/lib64/libdl.so
-- Looking for cblas_sgemm
-- Looking for cblas_sgemm - found
-- MKL library found
-- Found Tclsh: /bin/tclsh (found version "8.5") 
-- Found TCL: /usr/lib64/libtcl.so  
-- Found TCLTK: /usr/lib64/libtcl.so  
-- Found TK: /usr/lib64/libtk.so  
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found JPEG: /usr/lib64/libjpeg.so (found version "62") 
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED
-- Found boost_headers 1.70.0 at /usr/local/lib/cmake/boost_headers-1.70.0
-- Found Boost: /usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found version "1.70.0")  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Downloads/openflow/RePlAce/build
`

username on machine is different than 'user'

mgwoo, 2019-08-23:
DBU is defined in
https://github.com/The-OpenROAD-Project/RePlAce/blob/021e4b36670d8ff5be53b38b86797cfe954f5b22/module/flute/src/dbu.h#L32

Could you give me a log with "$ make VERBOSE=1"?

(I hope there exists a -I${FLUTE_HOME}/src in your commands.)

speedbooster, 2019-08-23:
> (I hope there exists a -I${FLUTE_HOME}/src in your commands.)
I am not issuing commands myself. I am just trying to build the tools downloaded from git as-is. I did, however, copy lef, def and flute to 'module' or 'third_party' folders where needed.

I just looked into **RePlAce/module** folder and there is no **src** folder in **module/flute**. Its a flat directory structure and no **dbu.h** file exists. However, there is a folder **module/flute_2_2** which contains **src/dbu.h**.

mgwoo, 2019-08-23:
It sounds really weird to me...
Could you newly clone the RePlAce directory with the following commands after cleaning your directory?

    $ git clone --recursive https://github.com/The-OpenROAD-Project/RePlAce

speedbooster, 2019-08-23:
Ok, i just realized my **RePlAce/module** contents are different from those of **git**.

speedbooster, 2019-08-23:
I think I may have renamed **flute** shipped with **RePlAce** to **flute_2_2** and copied the **flute** from its original source to **RePlAce/module** directory..

Which brings me to a question: Do I re-compile the sources for **lef**, **def** and **flute** for each flow component? For example, these common tools are used by **Resizer** and **RePlAce**, but each have their own subdirectories (**module** or **third_party**) which contain the sources to these common tools (lef, def and flute). Are these sources synced to their original repository?
Is it alright to install them as individual tools and set-up **$PATH** to allow Resizer and RePlAce to call them? Thanks.

closed 2019-08-25 by speedbooster