The-OpenROAD-Project/RePlAce #55: Segfaults on examples

foshardware, 2019-10-12:
There is a problem with the lef parser.

```
==8425== Memcheck, a memory error detector
==8425== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==8425== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==8425== Command: replace -bmflag etc -lef ispd18_test1/ispd18_test1.input.lef -def ispd18_test1/ispd18_test1.input.def -output . -dpflag NTU3 -dploc ../../ntuplace/
==8425== 
CMD :  replace -bmflag etc -lef ispd18_test1/ispd18_test1.input.lef -def ispd18_test1/ispd18_test1.input.def -output . -dpflag NTU3 -dploc ../../ntuplace/ 

[INFO] CompileDate = Oct 12 2019
[INFO] CompileTime = 16:45:42
[INFO] StartingTime = Sat Oct 12 16:47:49 2019

[INFO] TargetDensity = 1.000000
[INFO] ExperimentIndex = 10
[INFO] DirectoryPath = ./etc/ispd18_test1.input/experiment010
[PROC] Begin Importing Placement Input ...
==8425== Invalid read of size 8
==8425==    at 0x5048FC3: _Unwind_Resume (unwind.inc:240)
==8425==    by 0x325405: LefDefParser::lefiViaRule::lefiViaRule() [clone .cold] (lefiViaRule.cpp:404)
==8425==    by 0x38EDFD: unitsCB(LefDefParser::lefrCallbackType_e, LefDefParser::lefiUnits*, void*) (lefParser.cpp:1670)
==8425==    by 0x6A2D23: LefDefParser::lefyyparse() (lef.y:579)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425==  Address 0x52d81d0 is 8 bytes after a block of size 8 alloc'd
==8425==    at 0x483877F: malloc (vg_replace_malloc.c:309)
==8425==    by 0x686964: LefDefParser::lefiUnits::setDatabase(char const*, double) (lefiUnits.cpp:129)
==8425==    by 0x6ABEFA: LefDefParser::lefyyparse() (lef.y:625)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425== 
==8425== Invalid read of size 8
==8425==    at 0x5048767: _Unwind_RaiseException_Phase2 (unwind.inc:54)
==8425==    by 0x504901D: _Unwind_Resume (unwind.inc:241)
==8425==    by 0x325405: LefDefParser::lefiViaRule::lefiViaRule() [clone .cold] (lefiViaRule.cpp:404)
==8425==    by 0x38EDFD: unitsCB(LefDefParser::lefrCallbackType_e, LefDefParser::lefiUnits*, void*) (lefParser.cpp:1670)
==8425==    by 0x6A2D23: LefDefParser::lefyyparse() (lef.y:579)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425==  Address 0x52d81d8 is 16 bytes after a block of size 8 alloc'd
==8425==    at 0x483877F: malloc (vg_replace_malloc.c:309)
==8425==    by 0x686964: LefDefParser::lefiUnits::setDatabase(char const*, double) (lefiUnits.cpp:129)
==8425==    by 0x6ABEFA: LefDefParser::lefyyparse() (lef.y:625)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425== 
==8425== Invalid read of size 8
==8425==    at 0x5048FC3: _Unwind_Resume (unwind.inc:240)
==8425==    by 0x305F41: ~new_allocator (new_allocator.h:89)
==8425==    by 0x305F41: ~allocator (allocator.h:153)
==8425==    by 0x305F41: ~_Alloc_hider (basic_string.h:150)
==8425==    by 0x305F41: ~basic_string (basic_string.h:658)
==8425==    by 0x305F41: ParseLefDef() [clone .cold] (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425==  Address 0x52d81d0 is 8 bytes after a block of size 8 alloc'd
==8425==    at 0x483877F: malloc (vg_replace_malloc.c:309)
==8425==    by 0x686964: LefDefParser::lefiUnits::setDatabase(char const*, double) (lefiUnits.cpp:129)
==8425==    by 0x6ABEFA: LefDefParser::lefyyparse() (lef.y:625)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425== 
==8425== Invalid read of size 8
==8425==    at 0x5048767: _Unwind_RaiseException_Phase2 (unwind.inc:54)
==8425==    by 0x504901D: _Unwind_Resume (unwind.inc:241)
==8425==    by 0x305F41: ~new_allocator (new_allocator.h:89)
==8425==    by 0x305F41: ~allocator (allocator.h:153)
==8425==    by 0x305F41: ~_Alloc_hider (basic_string.h:150)
==8425==    by 0x305F41: ~basic_string (basic_string.h:658)
==8425==    by 0x305F41: ParseLefDef() [clone .cold] (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425==  Address 0x52d81d8 is 16 bytes after a block of size 8 alloc'd
==8425==    at 0x483877F: malloc (vg_replace_malloc.c:309)
==8425==    by 0x686964: LefDefParser::lefiUnits::setDatabase(char const*, double) (lefiUnits.cpp:129)
==8425==    by 0x6ABEFA: LefDefParser::lefyyparse() (lef.y:625)
==8425==    by 0x397B68: Replace::Circuit::ParseLef(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) (lefParser.cpp:3218)
==8425==    by 0x38B815: Init (lefdefIO.h:134)
==8425==    by 0x38B815: ParseLefDef() (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425== 
==8425== 
==8425== Process terminating with default action of signal 6 (SIGABRT): dumping core
==8425==    at 0x508DF25: raise (in /usr/lib/libc-2.30.so)
==8425==    by 0x5077896: abort (in /usr/lib/libc-2.30.so)
==8425==    by 0x503B0D3: _Unwind_Resume.cold (unwind.inc:245)
==8425==    by 0x305F41: ~new_allocator (new_allocator.h:89)
==8425==    by 0x305F41: ~allocator (allocator.h:153)
==8425==    by 0x305F41: ~_Alloc_hider (basic_string.h:150)
==8425==    by 0x305F41: ~basic_string (basic_string.h:658)
==8425==    by 0x305F41: ParseLefDef() [clone .cold] (lefdefIO.cpp:404)
==8425==    by 0x38BB41: ParseInput() (lefdefIO.cpp:239)
==8425==    by 0x329349: main (main.cpp:442)
==8425== 
==8425== HEAP SUMMARY:
==8425==     in use at exit: 174,738 bytes in 805 blocks
==8425==   total heap usage: 915 allocs, 110 frees, 459,211 bytes allocated
==8425== 
==8425== LEAK SUMMARY:
==8425==    definitely lost: 0 bytes in 0 blocks
==8425==    indirectly lost: 0 bytes in 0 blocks
==8425==      possibly lost: 0 bytes in 0 blocks
==8425==    still reachable: 174,738 bytes in 805 blocks
==8425==         suppressed: 0 bytes in 0 blocks
==8425== Rerun with --leak-check=full to see details of leaked memory
==8425== 
==8425== For lists of detected and suppressed errors, rerun with: -s
==8425== ERROR SUMMARY: 13 errors from 4 contexts (suppressed: 0 from 0)
```

foshardware, 2019-10-12:
`commit 021e4b36670d8ff5be53b38b86797cfe954f5b22`

mgwoo, 2019-10-24:
The newly developed version (in develop branch): 
https://github.com/The-OpenROAD-Project/RePlAce/tree/develop/test/testcases/ispd-test 
doesn't have this problem. Could you check this?

closed 2019-11-15 by mgwoo

mgwoo, 2019-11-15:
Close due to non-response...
This issue might be related to the new PR patch in lef/def repos.
https://github.com/The-OpenROAD-Project/lef/commit/4b20c5f8346832739617e3f47d3cf43bf8618d10
https://github.com/The-OpenROAD-Project/def/commit/bdf90b162727c56a6199c40f41a6ab50b9c97fb5