The-OpenROAD-Project/RePlAce #61: Build error on ubuntu 18.04

aolofsson, 2020-01-22:
What is the preferred branch for beta testers?

Following README, I get the following cmake error on build. (pre-requisites installed.)

=====================================================================

CMake Error at CMakeLists.txt:40 (add_subdirectory):
  The source directory

    /home/aolofsson/work/darpa/IDEA/ucsd-openroad/RePlAce/module/def

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:41 (add_subdirectory):
  The source directory

    /home/aolofsson/work/darpa/IDEA/ucsd-openroad/RePlAce/module/lef

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:43 (add_subdirectory):
  The source directory

    /home/aolofsson/work/darpa/IDEA/ucsd-openroad/RePlAce/module/OpenSTA

  does not contain a CMakeLists.txt file.

-- Could NOT find GOOGLEHASH (missing: GOOGLEHASH_INCLUDES GOOGLEHASH_COMPILE) 
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TCLTK (missing: TK_INCLUDE_PATH)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindTCL.cmake:213 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:52 (find_package)

mgwoo, 2020-01-22:
Hi Andreas, 

I guess submodules were not properly initialized on your machine.
Could you type the following command to clone all submodule repos for standalone mode? Cloning all submodules would help to solve the problem.

$ git clone --recursive https://github.com/The-OpenROAD-Project/RePlAce.git 

Kind Regards,

p.s. RePlAce's master branch is for standalone mode at this moment, but the master branch will have the integrated version of RePlAce.  Could you check the OpenROAD-flow repo, instead? (https://github.com/The-OpenROAD-Project/OpenROAD-flow). All of the OpenROAD tools exist within OpenROAD-flow repo.

aolofsson, 2020-01-22:
@mgwoo Thank you! Yes, this was a user error, I apologize for this! As you pointed out, one potential path us to use the docker approach is open-flow but it does have limitations in terms of modularity and development collaboration.

I do think there are a couple of things missing in the prerequisites..not sure how you/where you want to take the PR, so please see line below for ubuntu 18.04 (tk-dev, libjpeg..)

Here's what I used:

```bash
sudo apt-get install -y libx11-dev libboost-dev libcurl4 cmake swig flex libtool zlib1g-dev tcl-dev tk-dev libjpeg-dev
```
Builds now!

mgwoo, 2020-01-22:
Thanks for your comments. 
I'll also update the prerequisite/install_ubuntu18.sh on the RePlAce repo.

closed 2020-03-21 by mgwoo