The-OpenROAD-Project/RePlAce #62: negative HPWL

MartinGeisse, 2020-02-22:
The following design crashes replace (I just copied it in place of the gcd demo; I know that some other tool is going to choke on the missing clk signal later in the toolchain, but replace seems to have another problem):

module gcd(
  input[3:0] encoded,
  output reg[6:0] decoded
);
  always @(*) begin
    case (encoded)
      4'd0: decoded <= 7'b1110111;
      4'd1: decoded <= 7'b0100100;
      4'd2: decoded <= 7'b1011101;
      4'd3: decoded <= 7'b1101101;
      4'd4: decoded <= 7'b0101110;
      4'd5: decoded <= 7'b1101011;
      4'd6: decoded <= 7'b1111011;
      4'd7: decoded <= 7'b0100101;
      4'd8: decoded <= 7'b1111111;
      4'd9: decoded <= 7'b1101111;
      4'd10: decoded <= 7'b0111111;
      4'd11: decoded <= 7'b1111010;
      4'd12: decoded <= 7'b1010011;
      4'd13: decoded <= 7'b1111100;
      4'd14: decoded <= 7'b1011011;
      4'd15: decoded <= 7'b0011011;
      default: decoded <= 0;
    endcase
  end
endmodule

At some point, this outputs:

[INFO] Nesterov: 10 OverFlow: 0.5945 ScaledHpwl: 5583.1606
... (repeated many times, but the OverFlow and ScaledHpwl values don't change)
[INFO] Nesterov: 2100 OverFlow: 0.5945 ScaledHpwl: 5583.1606
[INFO] Timing: WNS = 1e+30
[INFO] Timing: TNS = 0
[INFO] Nesterov: 2110 OverFlow: 0.4528 ScaledHpwl: -40109.2656
... (repeated many times, but the OverFlow and ScaledHpwl values don't change)
[INFO] Nesterov: 2500 OverFlow: 0.4528 ScaledHpwl: -40109.2656
[INFO] Timing: WNS = 1e+30
[INFO] Timing: TNS = 0
NEGATIVE HPWL ERROR! _00_ 72.0000 587.4428
NEGATIVE HPWL ERROR! _00_ 79.2643 592.2643
NEGATIVE HPWL ERROR! 
_00_-80.1800 -79.8000

I have seen a commit that mentions the "negative HPWL" error, but the commit is stuck in the alpha2.1 branch. Is that commit relevant to the error I'm getting, and is the commit planned to be merged at some point in the future?

mgwoo, 2020-02-22:
That's a divergence problem in the previous RePlAce. Usually, very small designs like gcd are not converged well. The RePlAce engine recommends having at least 1K instances to place cells. I think you could tune the "-bin_grid_count" as higher(64, 128), or tune the -init_density_penalty(or -init_lambda_coeff in previous replace) parameter as higher(0.01, 0.1).

MartinGeisse, 2020-02-22:
Thanks for your quick reply. I suspected this to be a problem with small designs, but then couldn't reproduce that reliably. I'll try something larger for experiments instead.

MartinGeisse, 2020-02-23:
"-bin_grid_count 64" did the job. Thanks again!

closed 2020-02-23 by MartinGeisse