The-OpenROAD-Project/RePlAce #63: Support for non-rectangular standard-cell area - Need clarification

rbarzic, 2020-03-16:
Hello,

I have a design with a non-rectangular standard-cell area (The initial DEF has several macros placed and the std cells rows do no overlap with the macros)  - It crashes with a free() invalid pointer/core dump when running the rep init_replace command
When looking at the readme it mentions: 
RePlAce does not support rectilinear layout regions 
but the GIFs clearly show non-rectangular standard cell area (which is the definition of a rectilinear region for me) .
So I'm wondering if I should investigate more in debugging the issue or if Replace simply does not support non-rectangular area

mgwoo, 2020-03-16:
The "RePlAce does not support rectilinear layout regions" means that RePlAce can't understand the DEF syntax like "DIEAREA ( x1 y1 ) ( x2 y2 ) ( x3 y3 ) ... ( xn yn )".

The gif you've seen is that macros are given and those macros have "FIXED" attributes in DEF, which is different from rectilinear supports.

mgwoo, 2020-03-16:
If your target is want to have a blockage in macros like gif image, you can make a "DIEAREA ( lx ly ) ( ux uy )" statements in DEF and macrocells to have "FIXED" attributes in COMPONENTS section.
RePlAce also supports fragmented ROW (e.g. removing ROW definitions where std cells should not be placed). This ROW definition will enable you to make some additional halos around each macro. 

OpenROAD-flow has open macro design testcases. (e.g. nangate45-swerv_wrapper / nangate45-tinyRocket designs -- [Link](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/c89bf32a44ab81635687489d4dce6c0b4188763d/flow/Makefile#L15-L16)) You can also check the expected input DEF that have FIXED macros and fragmented-ROW testcases in the following link:
https://github.com/The-OpenROAD-Project/RePlAce/blob/openroad/test/medium05.def

FYI, RePlAce is cleanly rewritten and verified in the openroad branch with a new command (global_placement), but RePlAce in the openroad branch is not a standalone binary anymore. I have a plan to update the master branch to point the openroad branch.

rbarzic, 2020-03-16:
Thank you for the clarification !

closed 2020-03-16 by rbarzic