The-OpenROAD-Project/RePlAce #65: make[3]: don't know how to make /usr/ports/cad/replace/work/RePlAce-1.1.1-277-gcf289bb/src/replace/src/replace.i. Stop

yurivict, 2020-03-29:
You have
```
set(REPLACE_HOME ${PROJECT_SOURCE_DIR}/src/replace )
...
add_custom_command(OUTPUT ${REPLACE_WRAP}
  COMMAND ${SWIG_EXECUTABLE} -tcl8 -c++ -o ${REPLACE_WRAP} ${REPLACE_HOME}/src/replace.i
  WORKING_DIRECTORY ${REPLACE_HOME}
  DEPENDS ${REPLACE_HOME}/src/replace.i ${REPLACE_HOME}/include/replace/Replace.h
)
```
which looks for ```replace.i``` in ```${PROJECT_SOURCE_DIR}/src/replace/src/replace.i``` which doesn't exist.

1.1.1-277-gcf289bb

mgwoo, 2020-03-29:
Could you compile on the top-level app repo? (e.g. https://github.com/The-OpenROAD-Project/OpenROAD). I had updated it in the [README.md](https://github.com/The-OpenROAD-Project/RePlAce#how-to-download-and-build)

The standalone mode will not be supported anymore.

yurivict, 2020-03-29:
> The standalone mode will not be supported anymore.

qflow also suggests that ```replace``` can be used with it as a standalone tool.

mgwoo, 2020-03-29:
> Besides, isn't it just an obvious bug with a path that needs to be fixed?

This is not a bug in the top-level app.
${PROJECT_SOURCE_DIR} is OpenROAD source dir and has the RePlAce submodule in the OpenROAD/src/replace.

OpenROAD/src/replace/src/replace.i is correct path to point the replace.i

yurivict, 2020-03-29:
> This is not a bug in the top-level app.
> ${PROJECT_SOURCE_DIR} is OpenROAD source dir and has the RePlAce submodule in the OpenROAD/src/replace.

> OpenROAD/src/replace/src/replace.i is correct path to point the replace.i

You don't have to hardcode global paths in a project. The desired behavior can be achieved by setting a cmake variable from outside.