The-OpenROAD-Project/RePlAce #66: Building issues Ubuntu 18.04 OpenSTA TclEncode.tcl needed, then replace.tcl needed

psgff, 2020-04-27:
Following the instruction. Cloned from git recursively:

```
git clone --recursive https://github.com/The-OpenROAD-Project/RePlAce.git
cd RePlAce
mkdir build
cd build
cmake ..
make
```
First I had this:
```
make[2]: *** No rule to make target '../src/OpenSTA/etc/TclEncode.tcl', needed by 'ReplaceTclInitVar.cc'.  Stop.
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/replace.dir/all' failed
make[1]: *** [CMakeFiles/replace.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
```

Then I've cloned OpenSTA into RePlAce/src/ and tried `make` again:
```
make[2]: *** No rule to make target '../src/replace/src/replace.tcl', needed by 'ReplaceTclInitVar.cc'.  Stop.
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/replace.dir/all' failed
make[1]: *** [CMakeFiles/replace.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
```

psgff, 2020-04-28:
My bad, sorry. Haven't checked out from standalone branch. Nevertheless here comes an error:
```
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
```

Superstite, 2020-05-14:
> Following the instruction. Cloned from git recursively:
>
> ```
> git clone --recursive https://github.com/The-OpenROAD-Project/RePlAce.git
> cd RePlAce
> mkdir build
> cd build
> cmake ..
> make
> ```
>
> First I had [...]

Hello, I am facing the same issue. Can you tell how did you solve this. Thanks in advance.

psgff, 2020-05-15:
Hello! I haven't :( I needed this to get qFlow work. Partly sovled it. See https://github.com/RTimothyEdwards/qflow/issues/10

aashishgethub, 2020-05-21:
I am also having exactly same issue please somebody tell how to solve it?

aashishgethub, 2020-05-21:
I also want it to get qFlow work, i am in agreat need of having RePlAce, I installed OpenROAD successfully but installing RePlAce from inside /usr/local/OpenROAD/build/src/replace and from inside /usr/local/OpenROAD/src/replace/build shows the same error,
`make[2]: *** No rule to make target '../src/OpenSTA/etc/TclEncode.tcl', needed by 'ReplaceTclInitVar.cc'.  Stop.
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/replace.dir/all' failed
make[1]: *** [CMakeFiles/replace.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`
Please help me, to solve it
thanks in advance

mgwoo, 2020-05-21:
If you want to use standalone mode (old legacy version), please do the following:

    $ git clone --recursive --branch standalone https://github.com/The-OpenROAD-Project/RePlAce.git

I'll update README.md on the standalone branch.
FYI, the standalone will not be supported anymore. Please use the top-level OpenROAD-app repo to get recent RePlAce. (https://github.com/The-OpenROAD-Project/OpenROAD.git)

aashishgethub, 2020-05-22:
I have installed OpenROAD successfully, please tell me how to build RePlAce from it? because installing RePlAce from inside /usr/local/OpenROAD/build/src/replace and from inside /usr/local/OpenROAD/src/replace/build shows the same error.

mgwoo, 2020-05-22:
If you type "global_placement" command in the top-level app, that command will call RePlAce.

aashishgethub, 2020-05-22:
I understand that global_placement command in the top-level app (openroad) will call RePlAce but actually I need RePlAce for building 'Qflow' so I need a standalone like RePlAce command for bash terminal that can be found by Qflow and I can build Qflow.
Please help me, how can I acheive this.

aashishgethub, 2020-05-25:
Please help me to install RePlAce standalone successfully. I am in a greate need of it. Thanks.

rovinski, 2020-05-25:
Standalone RePlAce is deprecated and should not be used. We will not be supporting it. You should look at the [OpenROAD-flow global placement script](https://github.com/The-OpenROAD-Project/OpenROAD-flow/blob/openroad/flow/scripts/global_place.tcl). The script can be run from a shell using `openroad -exit [SCRIPT]`

closed 2020-05-25 by rovinski