The-OpenROAD-Project/RePlAce #70: Segmentation Violation

praharshapm, 2020-08-19:
Hi,
I am getting segmentation violation error while executing the rep place_cell_nesterov_place:
`[INFO] 80pCellArea = -nan
[INFO] FillerInit: TotalFillerArea = 821.911865
[INFO] FillerInit: NumFillerCells = -2147483648
[INFO] FillerInit: FillerCellArea = -nan
[INFO] FillerInit: FillerCellSize = (-nan, -nan)
[ERROR]: during executing: "replace < /openLANE_flow/scripts//replace_gp.tcl |& tee >&@stdout /openLANE_flow/designs/AMUX2_3V_top/runs/19-08_07-19//logs/placement/replace.log"
[ERROR]: Last 10 lines:
child killed: segmentation violation
`

mgwoo, 2020-08-19:
Hello,

Could you please upload your benchmarks/testcases that generate those errors?

praharshapm, 2020-08-19:
I am running a small design for an analog multiplexer using Openlane which uses the OpenROAD flow. I am attaching my verilog files and the def file produced during floorplanning
[AMUX2_3V.zip](https://github.com/The-OpenROAD-Project/RePlAce/files/5095553/AMUX2_3V.zip)

praharshapm, 2020-08-19:
Do tell if any other file is required

mgwoo, 2020-08-19:
It looks like this bug had patched before. 
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/replace/src/nesterovBase.cpp#L1160-L1165

Quick question:  
Could you specify the version that you're using? 
You may copy and paste the commit that OpenROAD top-level app prints at the beginning of binary execution.

praharshapm, 2020-08-19:
I am using RePlAce Version: 1.0.0

So, basically this error is arising because the design is small?
Exactly how can I resolve this issue?
where should I include these lines:
` // when #instances are too small,
  // extracts average values in whole ranges.
  if( minIdx == maxIdx ) {
    minIdx = 0;
    maxIdx = dxStor.size();
  }
`

rovinski, 2020-08-19:
@praharshapm I think the real issue here is that openlane is using an old version of RePlAce. You should go encourage them to use a newer version which contains the fix.

praharshapm, 2020-08-19:
okay. Thank you so much.

closed 2020-08-19 by praharshapm